zesage/motion_sensors #3: Difference between this and sensors

savy-91, 2021-02-10:
Hi! I am currently choosing which sensors package to use in a new application and I am confused as to what is the advantace of this package compared to the official sensors package?

I can see that here we also have the magnetometer, is there any other advantage?

zesage, 2021-03-12:
Based on the official sensor package, add the magnetometer and orientation, if you don’t need them, please use the official.

wwwdata, 2021-04-23:
Also, this package has the option of configuring the update interval. The official package is just taking the maximum. Which on Apple devices is quite a lot (60-100HZ) for accelerometer/gyroscope updates.

Zabadam, 2021-07-29:
> Based on the official sensor package

Even a [typo](https://github.com/zesage/motion_sensors/blob/master/lib/motion_sensors.dart#L107) in `UserAccelerometerEvent` constructor ;)

On a serious note, the additional features make this package a clear choice for me. Transitioning my own packages to `motion_sensors` now.

| motion_sensors | sensors_plus |
|:-:|:-:|
| AccelerometerEvent | AccelerometerEvent |
| UserAccelerometerEvent  | UserAccelerometerEvent |
| GyroscopeEvent  | GyroscopeEvent |
|  | MagnetometerEvent |
|  | OrientationEvent |
|  | AbsoluteOrientationEvent |
|  | ScreenOrientationEvent |

As well as the capability to determine sensor availability and change sampling rate:

```dart
static const int TYPE_ACCELEROMETER = 1;
static const int TYPE_MAGNETIC_FIELD = 2;
static const int TYPE_GYROSCOPE = 4;
static const int TYPE_USER_ACCELEROMETER = 10;
static const int TYPE_ORIENTATION = 15; //=TYPE_GAME_ROTATION_VECTOR
static const int TYPE_ABSOLUTE_ORIENTATION = 11; //=TYPE_ROTATION_VECTOR

/// Determines whether sensor is available.
Future<bool> isSensorAvailable(int sensorType) async {
  final available = await _methodChannel.invokeMethod('isSensorAvailable', sensorType);
  return available;
}

/// Change the update interval of sensor. The units are in microseconds.
Future setSensorUpdateInterval(int sensorType, int interval) async {
  await _methodChannel.invokeMethod('setSensorUpdateInterval', {"sensorType": sensorType, "interval": interval});
}
```
> ... invoked on singleton `motionSensors` after import.