zesage/motion_sensors #4: Write a file with accelerometer event data for an hour in flutter

rubana28, 2021-03-01:
I am trying to build an app which will 
    1)capture accelerometer, gyroscope data 
    2)write the data into a file for 1 hour and after 1 hour send that file to the server without pressing any button

Now, I tried to define the timer class to perform that 1 hour countdown. But It didn't show the result.

Can anyone suggest me how can I write the accelerometer data for an hour into a file?

rubana28, 2021-03-02:
@zesage Can you please tell me the meaning of this line "Duration.microsecondsPerSecond ~/ 60" ? what does it mean by microsecondsPerSecond ~/60?

zesage, 2021-03-12:
it's the update interval (in microseconds), which means receive data from the sensor 60 times per second.

rubana28, 2021-03-12:
> it's the update interval (in microseconds), which means receive data from the sensor 60 times per second.

Thanks for your reply. Can you please tell me how can we store this sensor data into a file for 1 hour? @zesage

0ttik, 2021-12-29:
> > it's the update interval (in microseconds), which means receive data from the sensor 60 times per second.
> 
> Thanks for your reply. Can you please tell me how can we store this sensor data into a file for 1 hour? @zesage

You should use some kind of database probably. For example, Hive or Sqflite.