zesage/motion_sensors #7: Basic feature error: Magnetometer Always show 0.0, 0.0, 0.0

debugerer, 2021-04-08:
When I run the example on my iPhone 8P with iOS 14.4.2, the Magnetometer Always show 0.0, 0.0, 0.0,
other data is updated quickly.
My runtime environment is:
-----------------------------
flutter doctor
Doctor summary (to see all details, run flutter doctor -v):
[✓] Flutter (Channel stable, 2.0.4, on macOS 11.1 20C69 darwin-x64, locale zh-Hans-CN)
[✓] Android toolchain - develop for Android devices (Android SDK version 29.0.3)
[✓] Xcode - develop for iOS and macOS
[✓] Chrome - develop for the web
[✓] Android Studio (version 4.1)
[✓] VS Code (version 1.55.0)
[✓] Connected device (2 available)

8P (mobile)   • ios            • iOS 14.4.2
Chrome (web) • chrome                                   • web-javascript • Google Chrome 89.0.4389.114
-----------------------------
Xcode 12.4 (latest)
iOS 14.4.2 (latest)
-----------------------------
Thank you for developing such a good plug-in. Could you test the example again? Maybe it is a new bug, thank you very much.

zesage, 2021-04-19:
I tested it on my iPhones and found no problems.  Try to remove the other sensors in the initState function, only use the magnetometer and test again.

debugerer, 2021-04-20:
Hi, I test it on another iPhone, it works. It turns out that the hardware of my iPhone 8Plus is broken.
Thank you very much! This issue can be close now!

closed 2021-04-20 by debugerer