zesage/motion_sensors #16: Support Kotlin Gradle plugin version 1.5.20 and higher

BenCherif, 2022-10-11:
What went wrong:
The Android Gradle plugin supports only Kotlin Gradle plugin version 1.5.20 and higher.
The following dependencies do not satisfy the required version:
project ':motion_sensors' -> org.jetbrains.kotlin:kotlin-gradle-plugin:1.3.50

alex27riva, 2022-11-14:
Same issue here

rlmax, 2023-01-29:
same issue

jaykshirsagar05, 2023-02-27:
Any workaround found?

Nimo11, 2023-03-02:
same issue

rlmax, 2023-03-02:
#17 this should fix it.

jaykshirsagar05, 2023-03-23:
I figured out this workaround, however, this needs to be done after every time you do flutter clean!

AnByronV, 2023-06-07:
Try using this update:

motion_sensors_updated:
    git:
      url: https://github.com/AnByronV/motion_sensors_updated.git

jacky610456, 2023-08-01:
error From this Plugin file： example/android/build.gradle 

ext.kotlin_version = '1.3.50'

gassuncaocosta, 2023-08-11:
File "example/android/build.gradle" was not fixed in #17.
This line (ext.kotlin_version = '1.3.50') is still causing errors when building apk.

Cykelstrom, 2024-03-14:
> What I did was the following. Install all the dependencies and once the download is finished, enter the motion sensor library and go directly to the gradle configuration on the Android platform and there I make some changes such as the kotlin version and the minimum version of the SDK and it works correctly for me. I hope this information is of some use to you.

Thanks, this worked for me. Much appreciated!

merajhasan88, 2024-05-11:
> What I did was the following. Install all the dependencies and once the download is finished, enter the motion sensor library and go directly to the gradle configuration on the Android platform and [...]

In ubuntu this can be done by going into `root/.pub-cache/hosted/pub.dev/motion_sensors-0.1.1/android` and adding a newer version of kotlin in build.gradle that matches the one in the exact same folder on this github and increasing minimum version of SDK. Works correctly afterwards.

darkmoon3d, 2024-06-12:
I'm having same trouble building this using Flutterflow https://community.flutterflow.io/ask-the-community/post/kotlin-gradle-compile-error-ZWJNZLkNX5k7PCW . Also tried downloading code and building using VSC with same error. 

Has there been any updates on how to fix this?