zesage/motion_sensors #25: Please update your pub.dev package

buttonsrtoys, 2023-10-24:
Thanks for posting this package to pub.dev. I see you are keeping it current, which is great! Can you do the community the great favour of updating your package on pub.dev? I and other would be indebted.

eokdev, 2023-11-08:
I agree! you have to update your package bro

itsmelaxman, 2024-09-14:
Hi! For now, you can use this directly from GitHub. I’ll keep the package updated and will publish it to pub.dev soon.

https://github.com/sajilopatro/motion_sensor

WalrusTheProud, 2025-01-06:
I appreciate your work on this package!

I do second the request to update your pub.dev page. The manual GitHub solution doesn't work when using packages that depend on this package; editing dependencies of those packages shouldn't be necessary, Using dependency overrides feel like an unnecessary workaround to simply updating a pub.dev package, and can itself cause gradle errors, as is the case for me. 

Any package that depends on this package is essentially bricked right now because of gradle errors.

biklas7, 2025-01-06:
Recently I found this fork and it's published on pub.dev of you want to check it out.
https://pub.dev/packages/dchs_motion_sensors

Seems to be working perfectly fine, at least for me!